freifunkMUC/site-ffm #8: Configure VLAN for mesh on LAN

fpletz, 2015-02-14:
One should be able to configure the VLAN that should be used for mesh on LAN.

This should also be integrated into upstream Gluon.

rotanid, 2018-01-20:
fyi, the Gluon Release v2018.1 will support VXLAN for Mesh-on-LAN and Mesh-on-WAN

LukeSiX, 2018-12-08:
even for meshing via VXLAN it should be possible to choose an 802.1q VLAN-ID on an LAN port

rotanid, 2018-12-08:
as far as i remember the problem was, that there's no generic way to implement it as not every device supports VLAN and if it is supported, the way to configure it is not the same for each device.
but as always, contributions are welcome :)

LukeSiX, 2018-12-10:
based on the usage of tons of different vendors and internal switch capabilities i can only agree to you. It would be a neat feature, but not really usable in the main tree. IF someome knows what VXLAN or Meshing is and really need vlan tagging, he can configure this via CLI. 

from my point of view i would like to cleanup our github and close this request

krombel, 2018-12-11:
no objection from my side

closed 2018-12-11 by LukeSiX