freifunkMUC/site-ffm #14: MTU für fastd

t-offline, 2015-03-09:
Hallo,

da wir nun auch IPv6 Adressen bei den gwtest*.ffmuc.net zurückliefern, sollten wir dann aber auch die MTU für IPv6 anpassen. Der Wert 1426 ist für IPv6 zu hoch. Es dürfen maximal 1406 sein.

Servus
Christian

fpletz, 2015-03-09:
Hast du dazu schon mal Tests gemacht? Batman zerteilt nämlich auch Pakete, weil es eine MTU von 1532 oder so erwartet. Vermutlich macht nämlich eine noch kleinere MTU noch mehr Sinn.

t-offline, 2015-03-10:
Mit dem Testen tue ich mir relativ schwer. Ich versuche zwar das eine oder das andere nachzustellen, aber z.z. wird ja viel geändert und da komm ich nicht nach. Leider gibt es noch viele andere Merkwürdigkeiten die mir das Leben schwer machen.
Nun bei der MTU denke ich erstmal wie folgt:
Da "behauptet" der fastd tap er "könne" 1426. Bei IPv4 stimmt das auch.
Wenn dann aber jemand via IPv6 rummacht, dann ist diese Behauptung leider eine Lüge. Der tap kann dann keine 1426 transportieren.

Würde er aber 1406 "behaupten" dann kann er das auch für uns (bat) machen.

Also es geht mir erstmal nur darum dem bat die "Wahrheit" über die "tap" Fähigkeiten zu sagen.

fpletz, 2015-03-24:
2015.2 wird ohne IPv6 fastd Connections ausgerollt.

t-offline, 2015-08-30:
Hi,
ich habe da mal was rauskopiert und etwas ergänzt:

```
###############################################################################
http://fastd.readthedocs.org/en/v17/manual/mtu.html
if possible the MTU should be chosed small enough so that IP fragmentation can be avoided.
Unlike OpenVPN, fastd doesn’t support fragmentation itself, but relies on the IP stack to fragment packets when necessary.

Guidelines
    The basic overhead of a fastd packet in TUN mode over IPv4 is 39 Bytes when only null crypto is used and 52 Bytes for all other crypto methods
    TAP mode needs 14 bytes more than TUN mode
    Tunneling over IPv6 needs 20 bytes more than IPv4

###
Your base MTU is 1492 (like most German DSL lines) and you want to use TAP mode over IPv4 with any crypto method:
    Choose 1426 bytes (1492 - 52 - 14).
###
Your base MTU is 1492 (like most German DSL lines) and you want to use TAP mode over IPv6 with any crypto method:
    Choose 1406 bytes (1492 - 52 - 14 - 20).
###
Your base MTU is 1420 (like most German DSL lines with SixXS[set MTU to 1420] PPPoE IPv4 UDP AYIYA) and you want to use TAP mode over IPv6 with any crypto method:
    Choose 1334 bytes (1420 - 52 - 14 - 20).
###
Conservative choice when you want to transfer IPv6 inside the tunnel(base MTU is 1366 or 1346 ???):
    Choose 1280 bytes (not relevant when you use batman-adv inside the tunnel as batman-adv will take care of the inner fragmentation!!!).
###
Conservative choice when you don’t know anything (but assume the base MTU is at least 1280 so IPv6 can be supported; eg. SixXS default MTU setting) and want to support tunnels over IPv4 and IPv6 in TAP mode with any crypto method:
    Choose 1194 bytes (1280 - 52 - 14 - 20).
###

###############################################################################
```

Servus
Christian

P.S.
https://github.com/freifunk-gluon/gluon/blob/11bffa59db923e976587cfa981ee4d95d1307e5b/patches/packages/routing/0003-batman-adv-decrease-maximum-fragment-size.patch
Dort wird also die MTU z. Z. zwangsweise auf maximal 1280 gesetzt.
Ob das aber so bleibt?

fpletz, 2015-08-30:
Und was willst du uns damit nun sagen? Es wurde mit 1280 als MTU genau wegen dem Patch getestet...

closed 2019-05-17 by awlx