freifunkMUC/site-ffm #18: Wifi channel should be configurable by user

fpletz, 2015-09-23:
#### Description
In the wifi settings in expert mode, the user should be able to configure the wifi channels for 2.4GHz and 5GHz. This should either be set to "default" (= site.conf setting) or "auto" and all available channels. The underlying wifi settings in uci (`/etc/config/network`) should be persistent and not be touched after sysupgrades if not set to "default".

#### Use Cases
  * Installations with meshing over wired connections and many access points in close proximity (due to many clients), it would make sense to disable wireless meshing and use the full wifi spectrum to increase usability.
  * Exclusive Mesh-Nodes without client wifi can be established to bridge distances where no cables can be installed and wouldn't impact regular users on the default channel.

#### Implementation
A bit of research is needed if this can be achieved with a new package or the existing `gluon-luci-wifi-config` packages need adjustments.

chris007de, 2015-09-23:
Some comments on this:
- Make sure that there are only the frequencies for indoor usage available in the 5Ghz band (Ch# < 100 afair, please check again), as long as OpenWRT is not ready for 5Ghz outdoor.
- To make debugging in case of "help! my nodes don't mesh!" more easily, I propose to add the channel and active network (client/mesh) settings to the distributed node information via Alfred page 159 (or whatever will substitute Alfred in the future).

Question: Is there also the plan to make "enable/disable of client/mesh networks" available to LuCi (if this is not already the case)?
I could image that advanced/expert user who want to change the channel also want to change the active networks.

fpletz, 2015-09-23:
@chris007de Yup, client and mesh wifi can already be toggled separately in the expert settings.

:+1: on your comments!

LukeSiX, 2018-12-08:
no need for changing this in LUCI, experts who wanna change that can use CLI   uci

closed 2018-12-08 by LukeSiX