freifunkMUC/site-ffm #19: site.conf Konsistenz Frage

t-offline, 2015-09-25:
hallo,

sollte nicht die site.conf der 3 Segmente jeweils so sein?:

```
#####
ffmuc:
#####

  prefix4 = '10.80.32.0/19',
  prefix6 = 'fdef:ffc0:4fff:0::/64',

  wifi24 = {
    ssid = 'muenchen.freifunk.net',
    channel = 6,
    htmode = 'HT20',
    mesh_ssid = 'mesh.ffmuc',
    mesh_bssid = '02:0E:8E:1E:61:17',
    mesh_mcast_rate = 12000,
  },

  wifi5 = {
    ssid = 'muenchen.freifunk.net',
    channel = 44,
    htmode = 'HT40+',
    mesh_ssid = 'mesh.ffmuc',
    mesh_bssid = '02:0E:8E:1E:61:17',
    mesh_mcast_rate = 12000,
  },

  next_node = {
    ip4 = '10.80.32.1',
    ip6 = 'fdef:ffc0:4fff:0::1',
    mac = '16:41:95:40:f7:dc',
  },

#####
ffmuc-welcome:
#####

  prefix4 = '10.80.64.0/19',
  prefix6 = 'fdef:ffc0:4fff:1::/64',

  wifi24 = {
    ssid = 'muenchen.freifunk.net',
    channel = 6,
    htmode = 'HT20',
    mesh_ssid = 'mesh.ffmuc.welcome',
    mesh_bssid = '02:0E:8E:1E:61:18',
    mesh_mcast_rate = 12000,
  },

  wifi5 = {
    ssid = 'muenchen.freifunk.net',
    channel = 44,
    htmode = 'HT40+',
    mesh_ssid = 'mesh.ffmuc.welcome',
    mesh_bssid = '02:0E:8E:1E:61:18',
    mesh_mcast_rate = 12000,
  },

  next_node = {
    ip4 = '10.80.64.1',
    ip6 = 'fdef:ffc0:4fff:1::1',
    mac = '16:41:95:40:f7:dc',
  },

#####
ffmuc-umland:
#####

  prefix4 = '10.80.96.0/19',
  prefix6 = 'fdef:ffc0:4fff:2::/64',

  wifi24 = {
    ssid = 'muenchen.freifunk.net',
    channel = 6,
    htmode = 'HT20',
    mesh_ssid = 'mesh.ffmuc.umland',
    mesh_bssid = '02:0E:8E:1E:61:19',
    mesh_mcast_rate = 12000,
  },

  wifi5 = {
    ssid = 'muenchen.freifunk.net',
    channel = 44,
    htmode = 'HT40+',
    mesh_ssid = 'mesh.ffmuc.umland',
    mesh_bssid = '02:0E:8E:1E:61:19',
    mesh_mcast_rate = 12000,
  },

  next_node = {
    ip4 = '10.80.96.1',
    ip6 = 'fdef:ffc0:4fff:2::1',
    mac = '16:41:95:40:f7:dc',
  },
```

Servus
Christian

fpletz, 2015-09-26:
Ein Diff wär einfacher gewesen…

closed 2015-09-26 by fpletz

t-offline, 2015-09-26:
da has Du recht...

Ein einsehbares "etckeeper" (o.ä.) der server und der gateways wäre auch manchmal einfacher... (Vorschlag von mir)

commit ce57afab (2015-09-26): Fix prefix4 & next_node.ip4, fixes #19