freifunkMUC/site-ffm #24: Integration des site-select packages

chris007de, 2015-09-28:
Habe mal das gluon-config-mode-site-select Paket (https://github.com/freifunk-gcp/v2015.1) in unseren aktuellen Master gebaut.

Damit kann man mehrere site.confs in eine Firmware integrieren und sie im Gluon Luci und per Kommandozeile umschalten.
[image]

Bis auf die bekannte Bindestrich-Allergie von lua gab es keine Probleme, allerdings musste ich den site-code von ffmuc-umland in ffmuc_umland aendern damit es reibungslos funktioniert.
Ausserdem gibt es Probleme beim kompilieren, wenn unsere custom gluon-ebtables-filter-... Pakete eingebaut werden sollen.
Es gibt allerdings noch einiges zu testen und fixen bevor man daran denken koennte das zu verwenden:

- [ ] Upgrade von aktueller gluon-ffmuc-v2015.5 Firmware auf Firmware mit site-select (site.conf muss nach Upgrade identisch mit ffmuc.conf sein)
- [ ] Upgrade von aktueller gluon-ffmuc-umland-v2015.6 Firmware auf Firmware mit site-select (site.conf muss nach Upgrade trotz Aenderung von des site-codes von "ffmuc-umland" auf "ffmuc_umland" identisch mit ffmuc_umland.conf sein)
- [ ] Was passiert bei einem Upgrade auf eine Firmware, bei der der aktuell ausgewaehlte site-code nicht mehr verfuegbar ist?

- [ ] Herausfinden, warum unsere gluon-ebtables-filter Pakete nicht mit eingebaut werden koennen und fixen

:beers:

chris007de, 2015-09-28:
Einige Testergebnisse:
* Basis: gluon-ffmuc-v2015.5 (ffmuc) <br>
  Sysupgrade: gluon-ffmuc-v2015.5-site-select (default=ffmuc, ffmuc_umland) <br>
  => ffmuc :+1: 
* Basis: gluon-ffmuc-umland-v2015.6 (ffmuc-umland) <br>
  Sysupgrade: gluon-ffmuc-v2015.5-site-select (default=ffmuc, ffmuc_umland) <br>
  => ffmuc :-1: 
* Basis: gluon-ffmuc-umland-v2015.6 (ffmuc-umland) <br>
  Sysupgrade: gluon-ffmuc-v2015.5-site-select (ffmuc, default=ffmuc_umland) <br>
  => ffmuc_umland :+1: 
* Basis: gluon-ffmuc-v2015.5-site-select (eingestellt auf ffmuc) <br>
  Sysupgrade: gluon-ffmuc-v2015.5-site-select (ffmuc, default=ffmuc_umland) <br>
  => ffmuc :+1: 
* Hin- und herwechseln zwischen ffmuc und ffmuc_umland mittels Luci config mode und uci ueber SSH: :+1:

fpletz, 2015-10-24:
Ist drin und funktioniert: https://github.com/freifunkMUC/site-ffm/commit/7f3333a684e48109130455a1be878fb8df468279

closed 2015-10-24 by fpletz