freifunkMUC/site-ffm #28: Neustartprobleme mit 2015.7-39

andz82, 2015-11-26:
Vmlt. bereits seit 2015.7-37, aber erst mit -39 aufgefallen:

Bei vmtl hoherer Last startem meine Router neu. Teilweise alle 2-5mim.

841v9: Bei einer VPN-Last von ca. 3-4MBit erfolgt ein Neustart. 
Grafna: https://ffmuc.net/stats/dashboard/db/node?var-node_id=e8de27588810 (mittlerweile abgeschaltet, siehe v.a. 24.11.-25.11.)

NanostationM2Loco: Bei mehr als 10 Clients mit etwas Durchsatz (kein VPN!) erfolgt ein Neustart.
Grafna: https://ffmuc.net/stats/dashboard/db/node?var-node_id=24a43ca6c785 (siehe v.a. seit 16.11.)

1043v2.1: Auch hier erfolgt ein Neustart sobald gefuehlt ein bisschen Last aufkommt. Kommt dann Teilweise nichtmal bis zum aufbau der VPN Verbindung vor dem Neustart
Grafna: https://ffmuc.net/stats/dashboard/db/node?var-node_id=30b5c23e9d34 (Siehe v.a. die vielen Luecken seit 20.11. hier bleibt der Router quasi in einer Boot-Loop)

Kann ich irgendwelche Spezifischen Tests durchfuehren?

andz82, 2015-11-26:
Nachtrag: Man sieht bei der Nanostation sehr schoen, dass kurz vor dem Neustart die Load > 2 geht und auch sonst immer wieder deutlich ueber 1 liegt.

Vlt noch zu erwaehnen, dass die Nodes alle MeshOnWAN machen und die NanostationM2 (nicht Loco) MeshOnWAN und MeshOnLAN

andz82, 2015-11-27:
Noch ein Nachtrag: Alle Nodes wurde frisch installiert und zurückgesetzt mit 'firstboot'. MeshOnLAN und MeshOnWAN wurden jeweils nur über die Configoberfläche konfiguriert. Es erfolgte außer bei andznet_07 kein Eingriff über die Konsole. Bei andznet_07 wurde ein VLAN hinzugefügt, dieser läuft jedoch anscheinend stabil (ist ein CPE210)

midzer, 2015-11-30:
Confirmed. Mein WDR3600 muss selbst neugestartet haben, weil der Router nur eine Uptime von 8 Tagen laut stats hat. Ich bin mir ziemlich sicher, dass ich seit Wochen nicht manuell resetet habe

chris007de, 2015-12-01:
@midzer der Neustart vor 8 Tagen wird wahrscheinlich der vom Update selbst gewesen sein.

Wenn man im meshviewer nach der 2015.7-39 filtert und sich die Uptimes ansieht, kann man erkennen dass der Grossteil der Knoten schon relativ lange laeuft bzw. durchgehend seit dem Update.
Einer meiner Knoten hatte in den ersten Tagen seit dem Update auch taeglich einen reboot, laeuft aber nun seit 5 Tagen durch, obwohl Traffic/Clientzahl in der Zeit sogar gestiegen ist.

Auch die Andznet Knoten laufen nun wohl seit 5 Tagen durch?

LukeSiX, 2018-12-08:
Kein weiteres Update mehr? alle gemeldeten Nodes existieren nicht mehr. Daher nun geschlossen

closed 2018-12-08 by LukeSiX