freifunkMUC/site-ffm #40: use better default mtu

rotanid, 2018-01-20:
quote from https://github.com/freifunk-gluon/gluon/pull/1210
"It is known that a fastd MTU of 1280 bytes will cause excessive fragmentation for full packets on networks that advertise to transport payloads of 1280 bytes, because it neglects the headers that are being added by the batman-adv encapsulation."
see this link for more info.
we discovered an increase of around 30% in maximum possible speed on each node when we migrated in summer 2017

LukeSiX, 2018-12-08:
note for ourself, internal usage:

Minimum MTU 1312 = 1280 Bytes (Minimum MTU IPv6) + 32 Bytes (BATMAN v15)

Optimum MTU 1370 = 1500 Bytes Ethernet -  4 Bytes VLAN Tagging (VDSL) - 8 Byte PPP - 32 Bytes DSL Tunneling Overhead (eg. L2TP used 32 Bytes) - 52 Bytes Fastd-Crypto - 14 Bytes TAP Mode - 20 Bytes Transport via IPv6

Maximum MTU 1410 = 1500 Bytes Ethernet - 4 Bytes VLAN Tagging - 52 Bytes Fastd-Crypto - 14 Bytes TAP Mode - 20 Bytes Transport via IPv6

rotanid, 2018-12-08:
https://gluon.readthedocs.io/en/latest/user/faq.html#what-is-a-good-mtu-on-the-mesh-vpn

maybe a bit more is possible, but the practical gain compared to 1312 will be minimal, as most of the gain is not having to split almost every package like it is today with 1280

LukeSiX, 2018-12-10:
1280 vs 1312 = +2.5%
1312 vs 1370 = +4.4%
1280 vs 1370 = +7.0% 
1280 vs. 1410 = +10.1%

i totally understand the MTU problematic and to have a setting which per default has to fit for almost any "broken" DSL and tunnel techniques. 

But if you have experienced network engineers who want to get the most power out of the installations and if you have the chance go get an native ethernet circuit, it might be interesting to choose out of 3 different MTU sizes?

If you know that you are not using IPv6 Sixxs or other tunnels for IP connectivity, why would you use then MTU 1310? It looks like that the rise from 1280 to 1310 is only a gain of 2.5% of packet size, but caused about 30% of speed gain, because of reducing the fragmentation. I don't know if there is any higher gain of speed/bandwidth if you would use 1280 vs 1370 (or 1410 for native installations).

Is there a source if someone already did these speed/CPU/fragmentation tests? How did you got the results of about 30% speed gain?

rotanid, 2019-01-02:
we measured the speed using iperf3 on the same device and same gateway using 1280 and 1312 and got the ca. 30% gain (for IPv6 connections inside the tunnel, outside IP version shouldnt matter)
there may be small further improvements with an even higher MTU, but as this can't be chosen automatically, the Gluon FAQ recommends a sane default which more an more communities are using.

krombel, 2019-03-29:
we will switch to MTU 1370 while shifting to new gateways

rotanid, 2019-04-02:
did you do measurements which showed substantial gains when doing 1370 in comparison to 1312?
deviating from a well-established standard value should be carefully considered ;)

awlx, 2019-05-17:
Switched MTU to 1406 on new GWs:
https://github.com/freifunkMUC/site-ffm/blob/experimental/domains/ffmuc_muc_cty.conf#L51

krombel, 2019-06-05:
as stated all new segments will be using a MTU of 1406.
As this new segments got released in v2019.0.3 and a switch to those segments is already setup I will close this for now.

closed 2019-06-05 by krombel