freifunkMUC/site-ffm #46: Include ebtables-source-filter

krombel, 2018-12-11:
To not route private networks to the broad network

krombel, 2019-03-18:
Implemented in 894453a809f562ac

krombel, 2019-03-18:
Reopen as IPv6 is not working

rotanid, 2019-03-18:
https://gluon.readthedocs.io/en/v2018.2.x/package/gluon-ebtables-source-filter.html
you need an additional config

rotanid, 2019-03-18:
i'm not entirely sure about it, but maybe you could specify your whole subnet as extra_prefixes6 in every domain. maybe @mweinelt can say whether this is an acceptable easier way than specifying each /64 sperately

krombel, 2019-03-18:
The question is: What would be the benefit of each config:
- Joined
  - People might roam between zones - isn't that filtered gateway-side? - so this is no point in that case
- Separately
  - You have a strict separation
  - As nodes will not forward it upstream => no rubbish from nodes roaming as they won't get a response either way

Is that right so far @rotanid?

rotanid, 2019-03-18:
you're probably right, but as i didn't use domains myself so far, i can't provide the perfect answer/solution

krombel, 2019-03-19:
We added extra_prefixes6 and now it seems to be working.
We will go for setup dedicated IPv6-Adresses for each domain for now.

The IPv6-Issue got fixed in 13dec0e17574860b37e895065ba266e10373d4c6

commit 64d4f9ce (2019-03-18): add extra_prefixes6 (fix #46)
commit 13dec0e1 (2019-03-18): add extra_prefixes6 (fix #46)
commit 262531dd (2019-03-28): add extra_prefixes6 (fix #46)
closed 2019-03-19 by krombel