freifunkMUC/site-ffm #50: XML-Verarbeitungsfehler: Nicht übereinstimmendes Tag. Erwartet: </div>

krombel, 2019-03-05:
winzerer meldet über [chat.ffmuc.net](https://chat.ffmuc.net/freifunk/pl/qitx7r4hmp8qfcrcf9phjb9hor):
"Beim Abspeichern der Konfig vor dem reboot kommt:"
```
Fehlermeldung: XML-Verarbeitungsfehler: Nicht übereinstimmendes Tag. Erwartet: </div>. Adresse: http://192.168.1.1/cgi-bin/config/wizard Zeile Nr. 36, Spalte 421: <p>Dein Knoten <em>freifunk-asdf</em> startet gerade neu und wird anschließend versuchen, sich mit anderen Freifunk-Knoten in seiner Nähe zu verbinden. [...]
```

rotanid, 2019-03-06:
Könnte ein XML-Fehler in der i18n oder in einem custom package sein, hatten so manche beim updaten, da die Browser da irgendwas strikter handhaben.
Sorry, dass ich es nicht genauer sagen kann...

krombel, 2019-03-19:
Es war tatsächlich ein Problem in i18n. Danke für den Tipp.
Fixed in 048248419eaf1b12b17e58858bc9ad86cc8cde8e

closed 2019-03-19 by krombel