freifunkMUC/site-ffm #52: all domains show as ffmuc on map.ffmuc.net

krombel, 2019-03-19:
We have 3 domains:
- ffmuc
- ffmuc_umland
- ffmuc_welcome

When running v2019.0-exp~15 only ffmuc is shown on map although the node is correctly configured and working.

As the only change seems to be the firmware we will check this first. When needed this will be shifted to the correct place

rotanid, 2019-03-19:
maybe the map software "meshviewer" needs to be updated to work with the new domain setup from Gluon?

ponschab, 2019-03-20:
gluon-neighbour-info reports:
`"system":{"site_code":"ffmuc","domain_code":"ffmuc_umland"}`

Maybe that can be easily fixed by replacing "site_code" with "domain_code" here:
https://github.com/freifunkMUC/meshviewer/blob/a637277b136a1e07ac79ac7f2858ec3fd03cbb76/lib/infobox/node.js#L177

ponschab, 2019-03-20:
See also: https://github.com/ffrgb/meshviewer/pull/268/commits/d6c5a3cb0229d754c45dcc6c30e86ee3c53eccac

krombel, 2019-03-29:
This seems to not be related to the firmware itself.
Thanks for pointing that out @rotanid and @ponschab.

closed 2019-03-29 by krombel