freifunkMUC/site-ffm #57: node prefers mesh-on-wlan over mesh-on-vxlan

krombel, 2019-05-23:
Useful links:
Freifunk-Franken reports sth similar: [Node on BATMAN_V cannot measure throughput and always reports 1Mb/s](
https://mantis.freifunk-franken.de/view.php?id=92)

This is currently observed on vxlan but it seems it is related to BATMAN_V support.
 
This will probably be fixed when https://github.com/freifunk-gluon/gluon/issues/1726 gets closed

awlx, 2019-05-23:
Fixbar durch:
`echo 100000 > /sys/class/net/vx_mesh_lan/batman_adv/throughput_override
echo 100000 > /sys/class/net/vx_mesh_wan/batman_adv/throughput_override`

krombel, 2019-05-24:
Forwarded upstream as it seems we are not the only ones affected: https://github.com/freifunk-gluon/gluon/issues/1728

krombel, 2019-06-05:
This got "fixed" in 083cf05f55ce512484f4e2f904f12f231bb08a66 by using patches of FreifunkChemnitz which set the throughput of wired connections statically to a high value.

This got released in v2019.0.3

Closing it (for now) as the issue is fixed for our network although a "real fix" is outstanding.

closed 2019-06-05 by krombel