freifunkMUC/site-ffm #62: Building from scratch fails because trying to delete non-existing gluon branch "patched" fails

jj-freifunk-muc, 2019-06-08:
Checked out a fresh, clean copy of tag v2019.0.4, then tried to build from scratch, but build fails almost at the beginning:

```
[user@node001 site-ffm.test2]$ make

#########################
# FFMUC Firmware build
# Building release v2019.0.4 for branch stable

make manifest
make[1]: Entering directory `/home/user/site-ffm.test2'
rm -rf output
(cd gluon-build \
        && git remote set-url origin https://github.com/freifunk-gluon/gluon.git \
        && git fetch origin \
        && git checkout -q v2018.2.1)
ln -sfT .. gluon-build/site
make gluon-patch
make[2]: Entering directory `/home/user/site-ffm.test2'
echo "Applying Patches ..."
Applying Patches ...
(cd gluon-build; git branch -D patched)
error: branch 'patched' not found.
make[2]: *** [gluon-patch] Error 1
make[2]: Leaving directory `/home/user/site-ffm.test2'
make[1]: *** [gluon-prepare] Error 2
make[1]: Leaving directory `/home/user/site-ffm.test2'
make: *** [all] Error 2
```

Manually creating a branch called "patched" in the repo inside the gluon-build directory works around the issue and allows the build to move forward.

awlx, 2019-06-08:
It's a known issue. 

We maybe should move from the old and often patched Makefile to a buildscript of another community.

awlx, 2020-01-16:
Fixed via https://github.com/freifunkMUC/site-ffm/commit/026f468adc6cc9f440e80280d39d615890da900d#diff-b67911656ef5d18c4ae36cb6741b7965

closed 2020-01-16 by awlx