freifunkMUC/site-ffm #86: simplify site.mk

txt-file, 2021-03-18:
Take a look at https://gitlab.com/FreifunkChemnitz/site-ffc/-/commit/fecab42d198685ae15f924c5e61aa1af22b9487c and https://gitlab.com/FreifunkChemnitz/site-ffc/-/commit/9bf94c4cc002c497ddac733cea861c892e746075

Just to give you ideas to get a shorter and more readable site.mk

krombel, 2021-03-18:
The part you are referring to is autogenerated by [contrib/genpkglist.py](https://github.com/freifunkMUC/site-ffm/blob/stable/contrib/genpkglist.py)

It might be possible to rewrite that by a bunch of `FLASH_4MB` or different vars but ATM our current setup is working and we can update the generator-script and with that our site.mk. Sadly we don't lack the time to change the working architecture on our own.

But thanks for the hint. Eventually someone finds time to update (and most importantly test) it. Feel free to take that part.
In general: If you want to contact us more directly you can find us in [our chat](https://chat.ffmuc.net/freifunk/channels/firmware).

awlx, 2021-05-24:
I will close this, as the file will stay auto-generated for now.

closed 2021-05-24 by awlx