freifunkMUC/site-ffm #89: Slow initial wireguard connection and gateway switch

goligo, 2021-05-01:
The current mechanism to setup the wireguard  connection and to switch to a different gateway in case of connection failure is unsatisfactory. The script is run once a minute and then uses a random delay of up to 60 seconds until it is started. So initially it may take up to two minutes, until a connection is established.

In case of a gateway failure, e.g. temporary loss of internet connection to one of our server locations, it is even worse - due to the random gateway selections, several tries to unavailable gateways can occur, before a working gateway is selected. Clients which are currently connected, will not automatically get a new IP address or use the new gateway, but rather try to use the old gateway. The DHCP lease time is set to 10 minutes, so it will take another 8 minutes, until clients will send a new DHCP request and eventually get a new IP address and gateway, which works again.

awlx, 2021-05-02:
This is to be handled with more logic in https://github.com/freifunkMUC/wgkex/issues/39 @goligo

goligo, 2021-05-02:
While I see how wgkex can help with the Gateway selection, I would expect something like a connection monitor, that permantently checks the connection status and traffic of the wireguard tunnel, could do much better than a cron job running once a minute.

commit 1a866d17 (2021-06-10): Fix initial slow connection on ipv4 or Ipv6 only networks

This fixes #89
closed 2021-06-10 by awlx