freifunkMUC/site-ffm #95: radv_filter_simple throws errors, when connection is not available

goligo, 2021-05-19:
```
Mon May 17 21:44:01 2021 daemon.err micrond[1772]: BusyBox v1.33.0 () multi-call binary.
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:
Mon May 17 21:44:01 2021 daemon.err micrond[1772]: Usage: grep [-HhnlLoqvsrRiwFE] [-m N] [-A/B/C N] PATTERN/-e PATTERN.../-f FILE [FILE]...
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:
Mon May 17 21:44:01 2021 daemon.err micrond[1772]: Search for PATTERN in FILEs (or stdin)
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:      -H      Add 'filename:' prefix
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:      -h      Do not add 'filename:' prefix
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:      -n      Add 'line_no:' prefix
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:      -l      Show only names of files that match
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:      -L      Show only names of files that don't match
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:      -c      Show only count of matching lines
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:      -o      Show only the matching part of line
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:      -q      Quiet. Return 0 if PATTERN is found, 1 otherwise
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:      -v      Select non-matching lines
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:      -s      Suppress open and read errors
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:      -r      Recurse
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:      -R      Recurse and dereference symlinks
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:      -i      Ignore case
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:      -w      Match whole words only
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:      -x      Match whole lines only
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:      -F      PATTERN is a literal (not regexp)
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:      -E      PATTERN is an extended regexp
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:      -m N    Match up to N times per file
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:      -A N    Print N lines of trailing context
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:      -B N    Print N lines of leading context
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:      -C N    Same as '-A N -B N'
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:      -e PTRN Pattern to match
Mon May 17 21:44:01 2021 daemon.err micrond[1772]:      -f FILE Read pattern from file
Mon May 17 21:44:01 2021 user.notice radv_filter_simple: Setting IPv6 Gateway to
Mon May 17 21:44:01 2021 daemon.err micrond[1772]: Problem with specified source mac '-j'.
Mon May 17 21:44:01 2021 user.notice radv_filter_simple: Setting gateway failed failling back to accept all
```

mayrstefan, 2023-03-06:
This looks like https://github.com/freifunkMUC/gluon-packages/issues/20 which I reported two weeks ago.

grische, 2023-03-06:
If this is the same, it should be fixed with those two commits

* https://github.com/freifunkMUC/site-ffm/commit/cc571eef4b32870e7a58c4adb5769a2f64552ceb
* https://github.com/freifunkMUC/site-ffm/commit/3a255ed6889c076f40f514aae30ba1dc1d3eb20e

And hence be part of any 2023.3.1 and later firmware release.

grische, 2023-11-20:
Closing as we are using a different package with https://github.com/freifunkMUC/site-ffm/releases/tag/v2023.11.1 and newer.

closed 2023-11-20 by grische