freifunkMUC/site-ffm #97: B.A.T.M.A.N. connection is one way

awlx, 2021-05-20:
For some reason after a DSL disconnect, B.A.T.M.A.N. on the node sees the gateways but not vice versa which leads to no client network at all.

Maybe it's related to #96.

awlx, 2021-05-20:
*@awlx attached a* [message](https://chat.ffmuc.net/_redirect/pl/cjaxsh7efjrex8jbetdioem1mw) *from @awlx*

```
root@aw-erx-wireguard:~# batctl gwl
[B.A.T.M.A.N. adv openwrt-2019.2-11, MainIF/MAC: primary0/4e:82:34:a7:fd:e3 (bat0/18:e8:29:ae:f3:f0 BATMAN_V)]
  Router            ( throughput) Next Hop          [outgoingIf]  Bandwidth
  f2:00:26:09:00:00 (       19.6) 36:51:9b:a8:18:d6 [  mesh-vpn]: 1024.0/1024.0 MBit
  f2:00:25:09:00:00 (       19.6) 36:51:9b:a8:18:d6 [  mesh-vpn]: 1024.0/1024.0 MBit
* f2:00:28:09:00:00 (     1000.0) 36:51:9b:a8:18:d6 [  mesh-vpn]: 1000.0/200.0 MBit
  f2:00:27:09:00:00 (       19.6) 36:51:9b:a8:18:d6 [  mesh-vpn]: 1000.0/200.0 MBit
```

awlx, 2021-05-20:
*@awlx attached a* [message](https://chat.ffmuc.net/_redirect/pl/p3z5f3yaxjnk9nkxaqdm4n5rko) *from @awlx*

```
root@aw-erx-wireguard:~# batctl ping f2:00:28:09:00:00
PING f2:00:28:09:00:00 (f2:00:28:09:00:00) 20(48) bytes of data
Reply from host f2:00:28:09:00:00 timed out
^C--- f2:00:28:09:00:00 ping statistics ---
1 packets transmitted, 0 received, 100% packet loss
rtt min/avg/max/mdev = 0.000/0.000/0.000/0.000 ms
root@aw-erx-wireguard:~# 

```

awlx, 2021-05-20:
*@awlx attached a* [message](https://chat.ffmuc.net/_redirect/pl/ycj7xfhqpig5bds7yzb1ndm9wy) *from @awlx*

```
22:10:36.667268 IP6 fe80::216:caff:febc:db35.37678 > fe80::23b:d2ff:fe95:967f.8472: OTV, flags [I] (0x08), overlay 0, instance 935867
4e:82:34:a7:fd:e7 > 36:51:9b:a8:18:d6, ethertype Unknown (0x4305), length 66: 
        0x0000:  440f 3200 461b 6535 8dfb 4e82 34a7 fde3  D.2.F.e5..N.4...
        0x0010:  0020 0000 0401 001c 12de 0003 ba7a 5aa4  .............zZ.
        0x0020:  8001 0000 f346 2783 8000 0000 c992 3ecd  .....F'.......>.
        0x0030:  0000 0000                                ....
```

goligo, 2021-05-21:
Issue is now reproducible: If checkuplink is run twice, so that the second run just starts before the tunnel is established again, so tries to trigger another reconnect, while the reconnect of the first run is still ongoing, the connection first seems to work, but BATMAN stops functioning after 10 minutes. As far the issue is understood, the BATMAN connection times out on the Gateway.

The issue is mitigated with
https://github.com/freifunkMUC/community-packages/commit/da5a63ac4c350630b40f63a613d414cf3bca1470 (to prevent parallel execution of checkuplink)
and
https://github.com/freifunkMUC/community-packages/commit/0049764390a4e9b055b0e29584b3c292683a9aaa (to detect the erroneous state and restart the tunnel)

closed 2021-05-21 by awlx