freifunkMUC/site-ffm #142: wgkex client: Add feature to get gateway from wgkex

awlx, 2021-08-09:
This will help to spread all nodes over all possible gateways as wgkex can then send the gateway with the least nodes to the node requesting a gateway.

This needs code on the wgkex side as well:
https://github.com/freifunkMUC/wgkex/issues/39

goligo, 2022-10-09:
While I understand the idea to receive the gateway to connect from the server, instead of having a hardcoded list of gateways baked into the firmware, I don't quite get what this has to do with the wireguard key exchange? This is what the abbreviation "wgkex" means, right?

Maybe it would make sense to find a better name (e.g. configbroker) which better resembles the purpose of this server side component? Receiving the Freifunk gateway to connect to doesn't have anything to do with exchanging wireguard keys.

awlx, 2022-10-09:
I see no problem with the name, the software just has the ability to do both. Exchange keys and reply back with a set of servers which received your request.

awlx, 2022-10-09:
@DasSkelett already started on the server site of things. This can also be an optional feature. And will be published under /api/v2 probably.

https://github.com/freifunkMUC/wgkex/pull/87

goligo, 2022-10-09:
Well to me it sounds like adding the feature to serve DNS requests to a HTTP server. Adding some completely unrelated feature to some server side component, just because it is already there. As I tried to explain above - gateway load balancing doesn´t have anything to do with exchanging wireguard keys.

awlx, 2022-10-09:
Wgkex was never meant to only be a keyexchange it was ever intended to be a broker for clients that's why the components are already named like that.

The reason it does only keyexchanges as of now was simply lack of time ;).

goligo, 2022-10-09:
This is a joke right - the abbreviation wgkex is for wireguard key exchange. That is why I was asking whether we shouldn´t change the name, if it does become a more generic service?

awlx, 2022-10-09:
See this draft from the very beginning of the project https://github.com/freifunkMUC/wgkex/pull/29

goligo, 2022-10-09:
Maybe the name was a bad choice from the early beginning?

awlx, 2022-10-09:
It's main job is still the wireguard keyexchange the only difference is in the future that it will be smart and only programs keys to Gateways with enough resources and tells the Client about that decision.

closed 2024-03-27 by grische