freifunkMUC/site-ffm #163: [Device support] TP-Link Archer C25 v1

GoliathLabs, 2021-11-04:
I have built a ffmuc firmware for this device. I would like to add a patch for this device manually, as it seems to work quite stable (but set it to broken due to memory usage of both WLAN chipsets), so we can update the node (https://map.ffmuc.net/#!/en/map/50c7bf3b3f62) in the future.

goligo, 2021-11-04:
What about submitting a PR, if you already have a patch, instead of opening an issue?

goligo, 2021-12-06:
Closing, as there is already firmware available for the Archer C25. It is not listed on https://firmware.ffmuc.net , but can be found in the firmware folders. 
http://firmware.ffmuc.net/stable/factory/gluon-ffmuc-v2021.11.1-tp-link-archer-c25-v1.bin
http://firmware.ffmuc.net/stable/sysupgrade/gluon-ffmuc-v2021.11.1-tp-link-archer-c25-v1-sysupgrade.bin

closed 2021-12-06 by goligo