freifunkMUC/site-ffm #169: Use gluon-mesh-vpn-wireguard package instead of own wireguard/vxlan implementation

goligo, 2022-02-09:
As Gluon will release wireguard/vxlan support with the next release, we should evaluate, whether we can use their implmentation instead of our current ffmuc-mesh-vpn-wireguard-vxlan package (using the checkuplink-script  to create wireguard tunnel and vxlan interface).

A first test of the package currently available in master and discussion in a Gluon meetup revealed following differences to our current approach:

- No support for automatic key propagation (broker URL) yet. However Gluon developers will look into it and/or accept PRs for adding this capability
- VNI is calculated differently (using domain_seed_bytes("gluon-mesh-vxlan", 3) instead of domain_seed_bytes("gluon-mesh-**vpn**-vxlan", 3)
- VXLAN local address is calculated differently (using printf instead of echo causing a different md5sum)
- VXLAN remote address is hardcoded to fe80::1
- Configuration format is different - using a map instead of an array for gateways and slightly different property names

So unfortunately, although some ideas from ffmuc made it into the implementation, it is highly incompatible and does not provide sufficient configuration options or extension mechanisms to make it compatible to our current gateway implementation. For the transition phase, we would need to be compatible for old and new wireguard clients, which leads to some interesting problems.

goligo, 2022-02-10:
In case anybody else is interested, I try to sum up some information regarding the current state of gluon-mesh-vpn-wireguard. Actually the package itself does not contain much and made we wonder where the real functionality is hidden:

https://github.com/freifunk-gluon/gluon/tree/master/package/gluon-mesh-vpn-wireguard

The creation and parameterization of the interfaces was put into gluon-mesh-vpn-core instead, in the netifd/proto/gluon_wireguard.sh

https://github.com/freifunk-gluon/gluon/tree/master/package/gluon-mesh-vpn-core/files/lib/netifd/proto/gluon_wireguard.sh

And the monitoring of the wireguard connection and switching of wireguard peers happens in the wgpeerselector

https://github.com/freifunk-gluon/packages/blob/master/net/wgpeerselector/files/usr/bin/wgpeerselector

In general the approach seems to be good - instead of destroying and recreating interfaces, the wireguard and vxlan interfaces are created by netifd once, and wgpeerselector just switches the peer as needed on the existing interface. This is can only be achieved by having a fixed vxlan remote ip address, as there seems to be no possibility to change the remote ip on an existing vxlan interface.

goligo, 2022-02-12:
After further discussion for us it seems to be a lot of effort on our gateway implementation, as the new package cannot just replace the current implementation on the client side, but we would need a complete set of new wireguard and vxlan interfaces on the gateways. On the other hand there are no benefits, apart from a shorter reconnection time in case of gateway failure.

@awlx stated she has no interested in investing time on this, I tend to agree. In case anybody else is interested, please reopen for further discussiom.

closed 2022-02-12 by goligo

maurerle, 2023-01-29:
Just as info:
another difference is the use of `udp6zerocsumtx udp6zerocsumrx` when setting up the vxlan interface in upstream gluon package in contrast to the ffmuc package.

I adjusted the ffmuc package so that it works with upstream `gluon-mesh-vpn-wireguard` here: https://github.com/maurerle/gluon-mesh-vpn-wireguard/tree/main/ff-mesh-vpn-wireguard-openwrt19