freifunkMUC/site-ffm #243: Add Segment Wertingen

istrator2, 2023-05-31:
In the next segment expansion _Wertingen_ please also create.

Hoping it will help with fundraising ;)

awlx, 2023-05-31:
Which SSID would you prefer? werting.freifunk.net (you probably also want the domain then) or something else?

istrator2, 2023-05-31:
I would appreciate **wertingen.freifunk.net** very much.

You are right. We would also like to have the domain - but we don't have much plans there. Just a _powered by Freie Netze München e.V._ and a redirect to ffmuc.net.

Who do I have to talk to, or does that fall right into the project with the TLS certificates?

awlx, 2023-05-31:
I think you somehow have to ask freifunk.net for the domain and then @DasSkelett and @GoliathLabs can add it. Maybe they can assist here :).

ce-4, 2023-05-31:
Seconding wertingen as segment name (werting would raise eyebrows), or maybe DLG for the whole Landkreis :)

istrator2, 2023-06-01:
Ich habe schon mal gefunden wo ich die Domain beantragen kann.
https://freifunk.net/kontakt/issuetracker/
Was möchte ich denn?

A-Record auf eine bestehende IP-Adresse
or
DNS-Record auf zwei bestehende DNS-Server für die Subdomain

istrator2, 2023-06-01:
> DNS-Record auf  anycast01.ffmuc.net
requested via issuetracker.

istrator2, 2023-09-28:
Bitte 2x /22 Netzsegmente anlegen.

Für die SSID in Spe wertingen.freifunk.net 
Eins für Normalbetrieb und eins für Events/Großinstallationen.

Siehe Chat @NOC 28.09.23 10:15

grische, 2023-10-23:
@awlx @GoliathLabs @DasSkelett do we already have a prefix that can be added to the firmware?

grische, 2023-10-31:
Following our previous IP numbering scheme for adding new domains, I added two new domains for wertingen in #266

Can someone review and implement this on the gateways?

closed 2023-12-05 by grische