freifunkMUC/site-ffm #388: gluon-info public key missing

T0biii, 2024-04-01:
when i run `gluon-info` on `v2024.3.1-next`(NanoPi R2S) the Public VPN Key does not show and i get this error
`sh: /lib/gluon/mesh-vpn/wireguard_pubkey.sh: Permission denied`
after making `chmod+x wireguard_pubkey.sh` the Public Key is shown

when i run it on v2024.3.1 (x86/64) i dont get the error but also i dont get the Public Key

grische, 2024-04-01:
We need to check if ou community package has the correct permissions (with chmod +x).

T0biii, 2024-04-01:
this is the mentioned file:
https://github.com/freifunk-gluon/community-packages/blob/master/ffmuc-mesh-vpn-wireguard-vxlan/files/lib/gluon/mesh-vpn/wireguard_pubkey.sh

grische, 2024-04-01:
Yeah, looks like it has wrong permissions.

closed 2024-04-01 by grische